pdx-robotics/bicopter-drone #17: Connect servo arm to pitch mechanism

Giandroid, 2019-03-29:
Use piano wires to connect the servo/arm to the pitch mechanism

jakechung42, 2019-04-10:
would fishing line work?

Giandroid, 2019-04-10:
I don't  think so. I think we need wire or a piano wire.

jakechung42, 2019-04-10:
do you have piano wire? I have violin wire, would that work?

Giandroid, 2019-04-10:
Is it stiff? It has to be able to both pull and push.

jakechung42, 2019-04-10:
it's not stiff. I'll bring it it so you can see

Giandroid, 2019-04-10:
I think paper clips might be best. I found this one at work- I will bring more.

We need to print the servo arms first though. It else there's nothing to attach the crossbars to. 
[image]