pdx-robotics/bicopter-drone #19: Program UART servos to work with PixHawk

Giandroid, 2019-03-29:
Verify that the digital servos can work with the usually-analog PixHawk flight controller

Chana030102, 2019-04-11:
After some discussion with @PrestonCazier on Discord, we've found that the LewanSoul LX-16A digital servos we'll be using will require a specific command sequence to control.

To interface them with the Pixhawk, we've found that we should [be able to specify certain servo ports to use a custom script](http://ardupilot.org/copter/docs/parameters.html#servo9-function).  The script can be [written in Python 2](http://ardupilot.org/planner/docs/using-python-scripts-in-mission-planner.html), so we'll want to develop a library to work with the commands for the servos.

To make sure we can continue using these servos with out buying a dedicated controller, we should test sending packets through an Arduino board. [Please be sure to follow the baud rate and other specifics noted in the command protocol pdf](https://github.com/pdx-robotics/bicopter-drone/blob/master/hardware/LewanSoul%20Bus%20Servo%20Communication%20Protocol.pdf)

Giandroid, 2019-04-12:
Awesome, can you make an issue to track developing the library?